arduino/include-robot #2: How to make the rubber part for the wheels

franva, 2022-11-15:
Hi there,

Thanks for sharing this interesting project and I am wondering that how are we going to make the rubber rim part for the wheels as 3D printer won't be able to print it?
[image]

csarnataro, 2022-11-15:
Hi, this part should be purchased as a separate part.
See for example:
https://www.pololu.com/product/4910

Anyway thank you for opening this issue, I will update the documentation with this information.

csarnataro, 2022-11-15:
BTW, we're working on the documentation these days, moreover we will add a few`.stl` files in the next few days. This project is still in a very early stage. Stay tuned!

closed 2022-11-15 by csarnataro