arduino/include-robot #17: Check if sleep when sending BLE command is still useful

csarnataro, 2024-01-12:


dido18, 2024-02-02:
I've reviewed the proposed changes, and after comparing both solutions, there is no need for the delay. 

- `Scratch3Arduino` use the delay https://github.com/lucarin91/include-robot/blob/d9c6cd7a81de92d240b13c1d00e43c06cf5d23a1/scratch-arduino-extensions/scratch-vm/src/extensions/scratch3_arduino/index.js#L1419
- `IncludeRobot` WITHOUT the delay  https://github.com/lucarin91/include-robot/blob/d9c6cd7a81de92d240b13c1d00e43c06cf5d23a1/scratch-arduino-extensions/scratch-vm/src/extensions/scratch3_arduino/index.js#L248

I tested with these two-scratch programs. Both can move the robot.
[image]

dido18, 2024-02-02:
As a side note: 
in both versions, the wheels aren't moving in sync since the messages are sent at different times. The need of a unique message is needed to ensure both wheels move together seamlessly.

closed 2024-02-02 by dido18