arduino/include-robot #29: Identify the right shield to use

csarnataro, 2024-01-12:
Choose which shield we should use (optionally more than one):
- Arduino custom shield
- Shield n.1 https://www.amazon.com/10pcs-XD-212-Multifunction-Expansion-arduino/dp/B0BYXYZF6W
- Shield n.2 https://wiki.dfrobot.com/Arduino_Nano_IO_Shield__SKU__DFR0012
- Other options?

umbynos, 2024-01-12:
Arduino custom shield:
CONS:
- Needs to be produced/open sourced
- The 3d design has to be modified (the board is larger)

PRO:
- has the ability to connect the power-switch
- The design is clever with connector on the top part

Shield N1 **~5 euro**:
CONS:
- needs a screw terminal for the battery connector
- bridge between VIN and 5V with soldering iron

PRO:
- we already have the 3d design (needs only a little fix for the barrel jack)

[Shield N2](https://wiki.dfrobot.com/Arduino_Nano_IO_Shield__SKU__DFR0012_) **~25 euro**:
CONS:
- The 3d design has to be modified (the board is larger)
- It's the most expensive one

PRO:
- with the use of a female-female jumper (VIN + / Power_Sel VCC) we don't have to use soldering iron to bridge stuff

dido18, 2024-05-24:
For the milestone 1, we are using the wiring on a breadboard